ballle98/AqualinkD #15: SPA and Pool Mode changes from AqualinkD

john1158way, 2019-04-23:
Hi Lee,    Thank you for your contributions to AqualinkD.  I am writing to ask if you have or plan to implement the ability to switch to modes (i.e SPA Drain, SPA Fill......) that are not available via All button but are available via One Touch and possibly the PDA.     User jsrichie opened an issue on the main AqualinkD page but I do not think it is going anywhere after Sean's comments.  

Thanks again
John

ballle98, 2019-04-23:
So I have a PDA only Jandy.  I looked into doing it but It does not seem to be an option available through the PDA.  The only way I know how to do it on my pool is but the system in service mode and press the button at the box.  If you know a way to tie it to AUX_6 or something similar that would be a good start.

john1158way, 2019-04-23:
Hi Lee,  thank you for responding.   I have done some investigation of the AUX method  Looking at JVA documentation the JVA has a common wire and two hot (24v) wires.  The valve position is determined by which of the wires is hot.   Right now I have SPA mode which recirculates the SPA and Pool mode which draws on the pool and splits the discharge between Pool Jets and the Spa with an overflow from the SPA creating a water feature.    I was thinking that I would use and AUX connection to to power an extra relay I have in my control box.  I would put this relay in the wiring path of the discharge valve actuator.   With the AUX off button the power to the valves would be routed as per normal but when the relay is pulled in the power is switched to opposite.  So if in Pool mode when the relay pulls in it switches the discharge to 100% SPA and I get SPA Fill,  If the mode is SPA then the discharge switches to Pool and I get SPA drain.   Albeit the slip stream into the SPA makes it less effective it will still drain.  I have tested by manually manipulating the valves in question.    What do you think?

john1158way, 2019-04-24:
found first blocker to my thinking  my relay does not have normally closed contacts only normally open.  would need to mount a different relay in box

ballle98, 2019-04-24:
you can look at the manual here https://www.jandy.com/-/media/zodiac/global/downloads/0748-91071/6594.pdf

john1158way, 2019-04-25:
Hi Lee,   thank you for the response.  I have the manual and the menu on page 39 shows one touch choices to change modes and looks to have some user definable modes.   (near top of menu breakdown on page 39)  that is what made me think it was possible to get commands that AqualinkD could send to make the change.     Sean has now replied and I answered his question so I apologize if you are getting duplicates on this topic if you follow his issues.