ballle98/AqualinkD #57: Unable to make on new Pi Zero

johntdyer, 2022-02-28:
Just downloaded on a new pi and seem to be getting an error I am unfamiliar w/. ( not the strongest C ) 

```
root@aqualinkd:/home/pi/AqualinkD# make
OS: 11 (bullseye)
GLIBC: ldd (Debian GLIBC 2.31-13+rpt2+rpi1+deb11u2) 2.31
./release/aqualinkd has been compiled
gcc -Wall -O3  -D AQ_PDA -D AQ_ONETOUCH -D AQ_IAQTOUCH -D AQ_RS16 -D MG_DISABLE_MD5 -D MG_DISABLE_HTTP_DIGEST_AUTH -D MG_DISABLE_MD5 -D MG_DISABLE_JSON_RPC  -o ./release/aqualinkd aqualinkd.o utils.o config.o aq_serial.o aq_panel.o aq_programmer.o net_services.o net_services_habridge.o json_messages.o rs_msg_utils.o devices_jandy.o packetLogger.o devices_pentair.o color_lights.o serialadapter.o mongoose.o timespec_subtract.o pda.o pda_menu.o pda_aq_programmer.o onetouch.o onetouch_aq_programmer.o iaqtouch.o iaqtouch_aq_programmer.o -l pthread -l m
/usr/bin/ld: onetouch.o:(.bss+0x0): multiple definition of `_menu'; pda_menu.o:(.bss+0x0): first defined here
collect2: error: ld returned 1 exit status
make: *** [Makefile:121: release/aqualinkd] Error 1
root@aqualinkd:/home/pi/AqualinkD#
```

```
root@aqualinkd:/home/pi/AqualinkD# uname -a
Linux aqualinkd 5.10.92+ #1514 Mon Jan 17 17:35:21 GMT 2022 armv6l GNU/Linux
root@aqualinkd:/home/pi/AqualinkD#
```

ballle98, 2022-02-28:
OK I can reproduce the issue.  It compiles fine on my older Pi 2B+ but on a newer Pi 4 64-bit it is not building

```
$ make
OS: 9 (stretch)
GLIBC: ldd (Debian GLIBC 2.24-11+deb9u4) 2.24
...
MD5 -D MG_DISABLE_HTTP_DIGEST_AUTH -D MG_DISABLE_MD5 -D MG_DISABLE_JSON_RPC  -c iaqtouch_aq_programmer.c  -o iaqtouch_aq_programmer.o
./release/aqualinkd has been compiled
gcc -Wall -O3  -D AQ_PDA -D AQ_ONETOUCH -D AQ_IAQTOUCH -D AQ_RS16 -D MG_DISABLE_MD5 -D MG_DISABLE_HTTP_DIGEST_AUTH -D MG_DISABLE_MD5 -D MG_DISABLE_JSON_RPC  -o ./release/aqualinkd aqualinkd.o utils.o config.o aq_serial.o aq_panel.o aq_programmer.o net_services.o net_services_habridge.o json_messages.o rs_msg_utils.o devices_jandy.o packetLogger.o devices_pentair.o color_lights.o serialadapter.o mongoose.o timespec_subtract.o pda.o pda_menu.o pda_aq_programmer.o onetouch.o onetouch_aq_programmer.o iaqtouch.o iaqtouch_aq_programmer.o -l pthread -l m
./release/aqualinkd has been compiled

```

Newer Pi 4

```
$ make
OS: 11 (bullseye)
GLIBC: ldd (Debian GLIBC 2.31-13+rpt2+rpi1+deb11u2) 2.31
...
/usr/bin/ld: onetouch.o:(.bss+0x0): multiple definition of `_menu'; pda_menu.o:(.bss+0x0): first defined here
collect2: error: ld returned 1 exit status
make: *** [Makefile:121: release/aqualinkd] Error 1

```

ballle98, 2022-02-28:
OK should be fixed now.  There are several string warnings I need to clean up but it compiles :)

johntdyer, 2022-03-01:
@ballle98  - Thanks for the quick turn around, pulled main down and now I'm getting the foloowing

```
pi@aqualinkd:~/AqualinkD $ sudo su -
cd root@aqualinkd:~# cd ~pi/AqualinkD
root@aqualinkd:/home/pi/AqualinkD# make clean
OS: 11 (bullseye)
GLIBC: ldd (Debian GLIBC 2.31-13+rpt2+rpi1+deb11u2) 2.31
rm -f *.o *~ ./release/aqualinkd  ./release/aqualinkd-player aq_serial_player.o ./release/aqualinkd-debug
rm -f aqualinkd.o utils.o  ./release/aqualinkd  ./release/aqualinkd-player aq_serial_player.o ./release/log_reader ./release/aqualinkd-player ./release/aqualinkd-debug
root@aqualinkd:/home/pi/AqualinkD# make
OS: 11 (bullseye)
GLIBC: ldd (Debian GLIBC 2.31-13+rpt2+rpi1+deb11u2) 2.31
gcc -Wall -O3  -D AQ_IAQTOUCH -D AQ_RS16 -D MG_DISABLE_MD5 -D MG_DISABLE_HTTP_DIGEST_AUTH -D MG_DISABLE_MD5 -D MG_DISABLE_JSON_RPC  -c aqualinkd.c  -o aqualinkd.o
gcc -Wall -O3  -D AQ_IAQTOUCH -D AQ_RS16 -D MG_DISABLE_MD5 -D MG_DISABLE_HTTP_DIGEST_AUTH -D MG_DISABLE_MD5 -D MG_DISABLE_JSON_RPC  -c utils.c  -o utils.o
gcc -Wall -O3  -D AQ_IAQTOUCH -D AQ_RS16 -D MG_DISABLE_MD5 -D MG_DISABLE_HTTP_DIGEST_AUTH -D MG_DISABLE_MD5 -D MG_DISABLE_JSON_RPC  -c config.c  -o config.o
config.c: In function ‘init_parameters’:
config.c:113:8: error: ‘struct aqconfig’ has no member named ‘pda_sleep_with_websock’
  113 |   parms->pda_sleep_with_websock = true;
      |        ^~
make: *** [Makefile:178: config.o] Error 1
root@aqualinkd:/home/pi/AqualinkD#
```

johntdyer, 2022-03-01:
Ignore that, I had mucked w/ the Makefile trying to get around the earlier bug.  Once I reset that file, make clean, and make'd I got much further :)