ballle98/AqualinkD #61: Web Interface is showing running when not?

heffneil, 2022-04-14:
Hey guys I am new to this but I got it working - except I have a few issues.  I did post in discussions but I don't know if anyone is reading those so I apologize for posting here as well:

1. Today my PDA only panel is showing me in the web interface it is running, however the pump is NOT running.

2. My time is wrong and I am unclear if / how I can update it to sync with my raspberry pi.  I do have the synch option enabled.

3. I don't know how to view / edit / run any programs within the web interface if at all possible.
Thanks!

ballle98, 2022-04-28:
Are you using my branch or the upstream one?  https://github.com/ballle98/AqualinkD/wiki#install-configuration

my version has support for PDA time update, but the https://github.com/sfeakes/AqualinkD upstream does not

Please go to the debug page http://[ip address]/debug.html (see https://github.com/sfeakes/AqualinkD#all-web-interfaces)

or log on the pi console (see https://github.com/ballle98/AqualinkD/wiki#run-in-foreground-with-logging)

heffneil, 2022-04-28:
I am using yours.  I did get some help where he said that I had a RS485 wiring issue.  I did have a loose ground so I fixed that but now I think its inconsident.  I am using yours because my panel is the PDA version but it never updated the time properly...

heffneil, 2022-04-28:
I have the debug file.  I removed some mqtt information, should I post it and anything else I should remove?

ballle98, 2022-04-28:
I don't think there is any personal information in the log.

heffneil, 2022-04-28:
it has mqtt password information I didn't want to post.  I didn't see anything else specific except IP addresses but that's not terribly important.  Below is the log:

Apr-17-22 11:42:48 AM Notice:  AqualinkD: Stopping!
Apr-17-22 11:42:48 AM Error:   RS Serial: Didn't find valid blocking serial port file descripter
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Exit!
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Aqualink Daemon v2.2.2
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Panel set to PDA-8 Combo Pool/Spa 
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config log_level         = 5
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config device_id         = 0x61
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config rssa_device_id    = 0x00
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config extra_device_id   = 0x00
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config extra_device_prog = NO
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config serial_port       = /dev/ttyUSB0
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config socket_port       = 80
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config web_directory     = /var/www/aqualinkd/
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config use_aux_labels    = NO
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config override frz prot = NO
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config mqtt_server       = 172.16.0.253:1883
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config mqtt_dz_sub_topic = domoticz/out
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config mqtt_dz_pub_topic = domoticz/in
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config mqtt_aq_topic     = aqualinkd
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config mqtt_user         = mqtt_user
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config mqtt_passwd       = 
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config mqtt_ID           = aqualinkd_bXXXXXX
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config idx water temp    = -999
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config idx pool temp     = -999
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config idx spa temp      = -999
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config idx SWG Percent   = 0
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config idx SWG PPM       = 0
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config PDA Mode          = YES
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config PDA Sleep Mode    = NO
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config force SWG         = NO
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config deamonize         = YES
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config log_file          = /var/log/aqualinkd.log
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config light_pgm_mode    = 0.00
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Debug RS485 protocol     = NO
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Display warnings in web  = NO
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Keep panel time in sync  = YES
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Read SWG direct          = NO
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Read ePump direct        = NO
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Read vsfPump direct      = NO
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Thread Network Services  = YES
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config BTN Filter_Pump   = label Filter Pump     | 
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config BTN Spa_Mode      = label Spa Mode        | 
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config BTN Aux_1         = label Cleaner         | 
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config BTN Aux_2         = label Waterfall       | 
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config BTN Aux_3         = label Spa Blower      | 
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config BTN Aux_4         = label Pool Light      | 
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config BTN Aux_5         = label Spa Light       | 
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config BTN Aux_6         = label NONE            | 
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config BTN Aux_7         = label NONE            | 
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config BTN Pool_Heater   = label Pool Heater     | 
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config BTN Spa_Heater    = label Spa Heater      | 
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Config BTN Solar_Heater  = label Solar Heater    | 
Apr-17-22 11:42:48 AM Notice:  NetService:Starting network services thread
Apr-17-22 11:42:48 AM Notice:  NetService:Starting web server on port 80
Apr-17-22 11:42:48 AM Notice:  NetService:Starting MQTT client to 172.16.0.253:1883
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Listening to Aqualink RS8 on serial port: /dev/ttyUSB0
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Waiting for Control Panel probe
Apr-17-22 11:42:48 AM Notice:  AqualinkD: Got message but no probe on '0x61', did we start too soon? (waiting for probe)
Apr-17-22 11:42:52 AM Notice:  AqualinkD: Starting communication with Control Panel
Apr-17-22 11:42:55 AM Notice:  Panl Prog: Programming: Init PDA, 10
Apr-17-22 11:42:55 AM Warning: RS Serial: Serial read bad Jandy checksum, ignoring
Apr-17-22 11:42:55 AM Warning: RS Serial: BAD PACKET To 0x61 of type PDA Init (*guess | HEX: 0x10|0x02|0x61|0x1b|0x01|0x00|0x8f|0x10|0xff|0x10|0x02|0x61|0x1b|0x01|0x00|0x8f|0x10|0x03|
Apr-17-22 11:42:55 AM Error:   AqualinkD: Nothing read on blocking serial port
Apr-17-22 11:42:55 AM Error:   AqualinkD: Aqualink daemon looks like serial error, resetting.
Apr-17-22 11:43:08 AM Notice:  Panl Prog: Programming: PDA init after wake, 15
[... cut ...]
Apr-28-22 14:19:43 PM Debug:   NetService:API: URI Request 'debug/download': value 0.00
Apr-28-22 14:19:43 PM Debug:   NetService:Served WEB request
Apr-28-22 14:19:43 PM Debug:   AqualinkD: RS received packet of type PDA Clear length 9
Apr-28-22 14:19:43 PM Debug:   PDA:       To 0x61 of type        PDA Clear | HEX: 0x10|0x02|0x61|0x09|0x00|0x00|0x7c|0x10|0x03|
Apr-28-22 14:19:43 PM Info:    NetService:URI request: '/api/debug/download '
Apr-28-22 14:19:43 PM Debug:   NetService:API: URI Request 'debug/download': value 0.00

ballle98, 2022-04-29:
This does not look like my fork of the code Are these the logs from the GUI?  Not sure how you captures the begining of the startup through debug.html I used the debug method on my wiki but you can also use "sudo journalctl -u aqualinkd".  There are several differences here that tell me this is not my version of the code:

1.  Version should be 2.2.2a
`AqualinkD: Aqualink Daemon v2.2.2a`

2. Not seeing all the "thread active" notifications like this one
`16:28:17.281 Debug:   Panl Prog: Thread 10,0xb5300f90 is active (Init PDA)`

3. You can see in my log where the date is captures on line 0 of the display
```
16:28:18.168 Debug:   PDA:       To 0x62 of type      Lng Message | HEX: 0x10|0x02|0x62|0x04|0x40|0x20|0x20|0x20|0x20|0x20|0x54|0x48|0x55|0x20|0x20|0x34|0x3a|0x32|0x37|0x50|0x4d|0xfd|0x10|0x03|
16:28:18.168 Debug:   PDA:       PDA Menu Line 0 =      THU  4:27PM

```

```
16:28:13.899 Notice:  AqualinkD: Aqualink Daemon v2.2.2a
16:28:13.905 Notice:  AqualinkD: Panel set to PDA-8 Combo Pool/Spa 
16:28:13.907 Notice:  AqualinkD: Config log_level         = 7
16:28:13.909 Notice:  AqualinkD: Config device_id         = 0x62
16:28:13.910 Notice:  AqualinkD: Config rssa_device_id    = 0x00
16:28:13.910 Notice:  AqualinkD: Config extra_device_id   = 0x00
16:28:13.912 Notice:  AqualinkD: Config extra_device_prog = NO
16:28:13.912 Notice:  AqualinkD: Config serial_port       = /dev/ttyUSB0
16:28:13.913 Notice:  AqualinkD: Config socket_port       = 8080
16:28:13.913 Notice:  AqualinkD: Config web_directory     = /var/www/aqualinkd/
16:28:13.913 Notice:  AqualinkD: Config use_aux_labels    = NO
16:28:13.913 Notice:  AqualinkD: Config override frz prot = NO
16:28:13.914 Notice:  AqualinkD: Config mqtt_server       = 192.168.0.95:1883
16:28:13.915 Notice:  AqualinkD: Config mqtt_dz_sub_topic = (null)
16:28:13.915 Notice:  AqualinkD: Config mqtt_dz_pub_topic = (null)
16:28:13.915 Notice:  AqualinkD: Config mqtt_aq_topic     = aqualinkd
16:28:13.916 Notice:  AqualinkD: Config mqtt_user         = (null)
16:28:13.917 Notice:  AqualinkD: Config mqtt_passwd       = (null)
16:28:13.917 Notice:  AqualinkD: Config mqtt_ID           = aqualinkd_b827eb6d
16:28:13.918 Notice:  AqualinkD: Config idx water temp    = -999
16:28:13.918 Notice:  AqualinkD: Config idx pool temp     = -999
16:28:13.919 Notice:  AqualinkD: Config idx spa temp      = -999
16:28:13.920 Notice:  AqualinkD: Config idx SWG Percent   = 0
16:28:13.921 Notice:  AqualinkD: Config idx SWG PPM       = 0
16:28:13.922 Notice:  AqualinkD: Config PDA Mode          = YES
16:28:13.922 Notice:  AqualinkD: Config PDA Sleep Mode    = YES
16:28:13.922 Notice:  AqualinkD: Config force SWG         = YES
16:28:13.922 Notice:  AqualinkD: Config habridge_server   = localhost:80
16:28:13.924 Notice:  AqualinkD: Config habridge_user     = habuser
16:28:13.924 Notice:  AqualinkD: Config deamonize         = NO
16:28:13.924 Notice:  AqualinkD: Config log_file          = (null)
16:28:13.926 Notice:  AqualinkD: Config light_pgm_mode    = 0.00
16:28:13.926 Notice:  AqualinkD: Debug RS485 protocol     = NO
16:28:13.926 Notice:  AqualinkD: Display warnings in web  = NO
16:28:13.927 Notice:  AqualinkD: Keep panel time in sync  = YES
16:28:13.927 Notice:  AqualinkD: Read SWG direct          = NO
16:28:13.928 Notice:  AqualinkD: Read ePump direct        = NO
16:28:13.928 Notice:  AqualinkD: Read vsfPump direct      = NO
16:28:13.929 Notice:  AqualinkD: Thread Network Services  = YES
16:28:13.930 Notice:  AqualinkD: Config BTN Filter_Pump   = label Filter Pump     | habid 1  
16:28:13.931 Notice:  AqualinkD: Config BTN Spa_Mode      = label Spa             | habid 2  
16:28:13.932 Notice:  AqualinkD: Config BTN Aux_1         = label Cleaner         | habid 3  
16:28:13.933 Notice:  AqualinkD: Config BTN Aux_2         = label NONE            | 
16:28:13.933 Notice:  AqualinkD: Config BTN Aux_3         = label NONE            | 
16:28:13.933 Notice:  AqualinkD: Config BTN Aux_4         = label Pool Light      | habid 4  
16:28:13.933 Notice:  AqualinkD: Config BTN Aux_5         = label Spa Light       | habid 5  
16:28:13.935 Notice:  AqualinkD: Config BTN Aux_6         = label NONE            | 
16:28:13.935 Notice:  AqualinkD: Config BTN Aux_7         = label NONE            | 
16:28:13.936 Notice:  AqualinkD: Config BTN Pool_Heater   = label Pool Heat       | habid 6  
16:28:13.937 Notice:  AqualinkD: Config BTN Spa_Heater    = label Spa Heat        | habid 7  
16:28:13.937 Notice:  AqualinkD: Config BTN Solar_Heater  = label NONE            | 
16:28:13.938 Notice:  NetService:Starting network services thread
16:28:13.944 Notice:  NetService:Starting web server on port 8080
16:28:13.951 Debug:   RS Serial: Openeded serial port /dev/ttyUSB0
[... 384 lines cut ...]
16:28:18.459 Debug:   PDA:       PDA Menu Line 3 = 
16:28:18.459 Debug:   PDA:       PDA Menu Line 4 = POOL MODE    OFF
16:28:18.459 Debug:   PDA:       PDA Menu Line 5 = POOL HEATER  OFF
16:28:18.461 Debug:   PDA:       PDA Menu Line 6 = SPA MODE     OFF
16:28:18.470 Debug:   PDA:       PDA Menu Line 7 = SPA HEATER   OFF
16:28:18.472 Debug:   PDA:       PDA Menu Line 8 = MENU            
16:28:18.472 Debug:   PDA:       PDA Menu Line 9 = EQUIPMENT ON/OFF
16:28:18.472 Debug:   PDA:       PDA Menu highlighted line = 4 = POOL MODE    OFF
16:28:18.473 Info:    PDA:       PDA Menu '2' Selectedline 'POOL MODE    OFF', Last line received ' 86`            '
16:28:18.474 Debug:   Panl Prog: Kicking PDA thread 10,0xb5300f90
[... cut ...]
16:28:19.415 Debug:   PDA:       To 0x62 of type           Status | HEX: 0x10|0x02|0x62|0x02|0x00|0x5f|0x73|0x08|0x00|0x50|0x10|0x03|
16:28:19.416 Debug:   Panl Prog: Kicking PDA thread 10,0xb5300f90
16:28:19.416 Debug:   PDA:       PDA can't sleep as thread 10,0xb5300f90 is active
16:28:19.417 Info:    Panl Prog: sent '0x00' to controller
16:28:19.418 Debug:   PDA:       waitForPDAMessageHighlight index 9
16:28:19.418 Debug:   PDA:       waitForPDAMessageHighlight last = 0x02 : index 4 : (1 of 10)
16:28:19.445 Debug:   AqualinkD: RS received packet of type PDA Hlight length 10

```

heffneil, 2022-05-01:
● aqualinkd.service - Aqualink RS daemon
     Loaded: loaded (/etc/systemd/system/aqualinkd.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: signal) since Sun 2022-05-01 11:23:23 EDT; 512ms ago
    Process: 27566 ExecStart=/usr/local/bin/aqualinkd $OPTS (code=killed, signal=SEGV)

ballle98, 2022-05-02:
in the directory where you did the git clone capture the output of these commands:

git remote -v
git log -1
sudo make install

ballle98, 2022-05-02:
make sure that you do the full procedure to install the master branch that is here.  https://github.com/ballle98/AqualinkD/wiki#install

The make clean is important because the committed bin file release/aqualinkd is not maintained

```
sudo apt-get install git
cd ~
mkdir git
cd git
git clone https://github.com/ballle98/AqualinkD.git
make clean
make
sudo make install
```

heffneil, 2022-05-02:
I did this and I think it is running.  Now I need to go back up and read above to see if I have the right version running now :) Thanks for all the help!

ballle98, 2022-05-03:
You have to do the programming with the PDA there is currently no way to do that with AqualinkD.  Good news is programs do not change when the battery dies and power is lost.  The battery is only used to keep time.  If you ever loose he PDA you can stop aqualinkd and use https://github.com/earlephilhower/aquaweb as a PDA replacement.  Another thing you can do is delete the programs and use Alexa or Google routines

aqualinkd <-> MQTT <-> openHAB <-> Alexa or Google

heffneil, 2022-05-03:
Yeah I thought about using my HA system to start and stop the system.  I had a weird thing happen where the web interface showed the pump running.  It wasn't.  I rebooted the pi and it shows it as off now.  I did do a debug but not sure what to check but concerning it's showing the wrong status?

heffneil, 2022-05-03:
> You have to do the programming with the PDA there is currently no way to do that with AqualinkD. Good news is programs do not change when the battery dies and power is lost. The battery is only used [...]

If there is a program that allows us to program via the web - is there a reason similar code couldn't be implemented to allow programming with aqualinkD?  Just wondering how it works differently Thanks again for all the help!

ballle98, 2022-05-03:
You should turn on the debug logs as before.  When you first open the browser AqualinkD has to refresh the states.  This takes about 7 seconds.  It then polls the status every 30 seconds.  This allows an opportunity for the PDA to get control.  Only one device can have control at a time.  If the PDA is active the web GUI can get out of sync until the PDA goes to sleep.

heffneil, 2022-05-03:
Ok the PDA hasn't been on for days.  This was showing running all day but it wasn't so I don't know whats going on there,.  Seems strange.

ballle98, 2022-05-03:
We should move this out of a bug report into a discussion thread but aqualinkd does translation where aquaweb does not.  Aqualinkd is much smarter than aquaweb.  You tell aqualinkd to turn on the filter pump and it does all the navigation for you.  With aquaweb you have to do all the button presses yourself and you interpret the output.  There are very complicated routines in Aqualinkd to set freeze protect and spa and pool temperature.

heffneil, 2022-05-03:
Im happy to chat whereever.   Much appreciated.

ballle98, 2022-05-03:
OK if you your still seeing an issue with the web interface running when it should not be then it's the same issue.  For things that are just questions we should use https://github.com/ballle98/AqualinkD/discussions

do a screen capture of the web interface and the debug logs

ballle98, 2022-05-03:
The point in you logs where it spits out Found Status for Filter Pump = 'FILTER PUMP' over and over again seems broken.  There may be something wrong with your config.  

This should not cause the issue but you should make sure that your PDA and aqualinkd don't have the same id.  Not sure if the PDA is 0x60 or 0x61.  If you are capturing logs and you use the PDA you'll see it show up.

```
Apr-28-22 14:19:43 PM Debug: PDA: To 0x61 of type Status | HEX: 0x10|0x02|0x61|0x02|0x00|0x00|0x00|0x00|0x00|0x75|0x10|0x03|
Apr-28-22 14:19:43 PM Debug: PDA: *** Found Status for Filter Pump = 'FILTER PUMP'
Apr-28-22 14:19:43 PM Debug: AqualinkD: RS received packet of type Status length 12

```

ballle98, 2022-05-05:
There is something going wrong here

This is what is in your logs:

```
Apr-28-22 14:19:38 PM Info: PDA: PDA Menu '20' Selectedline '-', Last line received 'EQUIPMENT STATUS'
Apr-28-22 14:19:38 PM Debug: AqualinkD: RS received packet of type Lng Message length 24
Apr-28-22 14:19:38 PM Debug: PDA: To 0x61 of type Lng Message | HEX: 0x10|0x02|0x61|0x04|0x02|0x20|0x20|0x46|0x49|0x4c|0x54|0x45|0x52|0x20|0x50|0x55|0x4d|0x50|0x20|0x20|0x20|0x41|0x10|0x03|
Apr-28-22 14:19:38 PM Debug: PDA: PDA Menu Line 0 = EQUIPMENT STATUS
Apr-28-22 14:19:38 PM Debug: PDA: PDA Menu Line 1 =
Apr-28-22 14:19:38 PM Debug: PDA: PDA Menu Line 2 = FILTER PUMP
Apr-28-22 14:19:38 PM Debug: PDA: PDA Menu Line 3 =
Apr-28-22 14:19:38 PM Debug: PDA: PDA Menu Line 4 =
Apr-28-22 14:19:38 PM Debug: PDA: PDA Menu Line 5 =
Apr-28-22 14:19:38 PM Debug: PDA: PDA Menu Line 6 =
Apr-28-22 14:19:38 PM Debug: PDA: PDA Menu Line 7 =
Apr-28-22 14:19:38 PM Debug: PDA: PDA Menu Line 8 =
Apr-28-22 14:19:38 PM Debug: PDA: PDA Menu Line 9 =
Apr-28-22 14:19:38 PM Info: PDA: PDA Menu '20' Selectedline '-', Last line received ' FILTER PUMP '
Apr-28-22 14:19:38 PM Debug: AqualinkD: RS received packet of type Status length 12
```

This is what is in my logs

```
16:28:19.861 Debug:   AqualinkD: RS received packet of type Lng Message length 24
16:28:19.862 Debug:   PDA:       To 0x62 of type      Lng Message | HEX: 0x10|0x02|0x62|0x04|0x00|0x20|0x20|0x20|0x45|0x51|0x55|0x49|0x50|0x4d|0x45|0x4e|0x54|0x20|0x20|0x20|0x20|0x10|0x10|0x03|
16:28:19.863 Debug:   PDA:       PDA Menu Line 0 =    EQUIPMENT    
16:28:19.863 Debug:   PDA:       PDA Menu Line 1 = 
16:28:19.864 Debug:   PDA:       PDA Menu Line 2 = 
16:28:19.865 Debug:   PDA:       PDA Menu Line 3 = 
16:28:19.865 Debug:   PDA:       PDA Menu Line 4 = 
16:28:19.867 Debug:   PDA:       PDA Menu Line 5 = 
16:28:19.867 Debug:   PDA:       PDA Menu Line 6 = 
16:28:19.867 Debug:   PDA:       PDA Menu Line 7 = 
16:28:19.868 Debug:   PDA:       PDA Menu Line 8 = 
16:28:19.869 Debug:   PDA:       PDA Menu Line 9 = 
16:28:19.869 Debug:   PDA:       *** Checking Equiptment '   EQUIPMENT'
16:28:19.871 Info:    PDA:       PDA Menu '19' Selectedline '-', Last line received '   EQUIPMENT    '
16:28:19.871 Debug:   Panl Prog: Kicking PDA thread 10,0xb5300f90
16:28:19.871 Debug:   PDA:       waitForPDAMessageTypesOrMenu last message type 0x04 (1)
16:28:19.872 Debug:   PDA:       PDA can't sleep as thread 10,0xb5300f90 is active
16:28:19.925 Debug:   AqualinkD: RS received packet of type Lng Message length 24
16:28:19.927 Debug:   PDA:       To 0x62 of type      Lng Message | HEX: 0x10|0x02|0x62|0x04|0x01|0x46|0x49|0x4c|0x54|0x45|0x52|0x20|0x50|0x55|0x4d|0x50|0x20|0x20|0x4f|0x46|0x46|0xbc|0x10|0x03|
16:28:19.928 Debug:   PDA:       PDA Menu Line 0 =    EQUIPMENT    
16:28:19.928 Debug:   PDA:       PDA Menu Line 1 = FILTER PUMP  OFF
16:28:19.929 Debug:   PDA:       PDA Menu Line 2 = 
16:28:19.929 Debug:   PDA:       PDA Menu Line 3 = 
16:28:19.931 Debug:   PDA:       PDA Menu Line 4 = 
16:28:19.931 Debug:   PDA:       PDA Menu Line 5 = 
16:28:19.931 Debug:   PDA:       PDA Menu Line 6 = 
16:28:19.932 Debug:   PDA:       PDA Menu Line 7 = 
16:28:19.933 Debug:   PDA:       PDA Menu Line 8 = 
16:28:19.933 Debug:   PDA:       PDA Menu Line 9 = 
16:28:19.935 Debug:   PDA:       *** Checking Equiptment 'FILTER PUMP '
16:28:19.935 Debug:   PDA:       *** Found EQ CTL Status for Filter Pump = 'FILTER PUMP  OFF'
16:28:19.935 Debug:   PDA:       set_pda_led from 4 to 1
16:28:19.936 Debug:   JandyDvce: Set SWG to off
16:28:19.937 Info:    PDA:       PDA Menu '19' Selectedline '-', Last line received 'FILTER PUMP  OFF'
16:28:19.937 Debug:   Panl Prog: Kicking PDA thread 10,0xb5300f90

```

Notice how on your Line 0 on yours is left justified but it is centered on mine.  Also notice how "PDA Menu Line 1 = FILTER PUMP  OFF" comes as one message on mine and yours is only "Menu Line 2 = FILTER PUMP" with no ON/OFF designation.  This is probably the source of the issue.

heffneil, 2022-05-05:
any recommendations on how to correct this?  the address for the pi shouldn't be an issue if I don't use the PDA?  Is that correct?  Let me know I clearly have some config issue here.

heffneil, 2022-05-05:
I changed my address on the pi and I enabled debug logging.  This is what I have at the bottom of my logs and this portion seems to be repeating a good bit:

May-05-22 11:47:21 AM Debug:   AqualinkD: RS received packet of type Lng Message length 24
May-05-22 11:47:21 AM Debug:   PDA:       To 0x63 of type      Lng Message | HEX: 0x10|0x02|0x63|0x04|0x82|0x20|0x37|0x31|0x60|0x20|0x20|0x20|0x20|0x20|0x37|0x34|0x60|0x20|0x20|0x20|0x20|0xce|0x10|0x03|
May-05-22 11:47:21 AM Debug:   PDA:       PDA Menu Line 0 =      THU 11:46AM
May-05-22 11:47:21 AM Debug:   PDA:       PDA Menu Line 1 = AIR         POOL
May-05-22 11:47:21 AM Debug:   PDA:       PDA Menu Line 2 =  71`     74`
May-05-22 11:47:21 AM Debug:   PDA:       PDA Menu Line 3 =
May-05-22 11:47:21 AM Debug:   PDA:       PDA Menu Line 4 = POOL MODE     ON
May-05-22 11:47:21 AM Debug:   PDA:       PDA Menu Line 5 = POOL HEATER  OFF
May-05-22 11:47:21 AM Debug:   PDA:       PDA Menu Line 6 = SPA MODE     OFF
May-05-22 11:47:21 AM Debug:   PDA:       PDA Menu Line 7 = SPA HEATER   OFF
May-05-22 11:47:21 AM Debug:   PDA:       PDA Menu Line 8 = MENU
May-05-22 11:47:21 AM Debug:   PDA:       PDA Menu Line 9 = EQUIPMENT ON/OFF
May-05-22 11:47:21 AM Debug:   PDA:       PDA Menu highlighted line = 4 = POOL MODE     ON
May-05-22 11:47:21 AM Info:    PDA:       PDA Menu '2' Selectedline 'POOL MODE     ON', Last line received ' 71`     74`    '
May-05-22 11:47:22 AM Debug:   AqualinkD: RS received packet of type Status length 12
May-05-22 11:47:22 AM Debug:   PDA:       To 0x63 of type           Status | HEX: 0x10|0x02|0x63|0x02|0x00|0x00|0x00|0x00|0x00|0x77|0x10|0x03|
May-05-22 11:47:22 AM Debug:   AqualinkD: RS received packet of type Status length 12
May-05-22 11:47:22 AM Debug:   PDA:       To 0x63 of type           Status | HEX: 0x10|0x02|0x63|0x02|0x00|0x00|0x00|0x00|0x00|0x77|0x10|0x03|
May-05-22 11:47:23 AM Debug:   AqualinkD: RS received packet of type Status length 12
May-05-22 11:47:23 AM Debug:   PDA:       To 0x63 of type           Status | HEX: 0x10|0x02|0x63|0x02|0x00|0x00|0x00|0x00|0x00|0x77|0x10|0x03|
May-05-22 11:47:23 AM Debug:   AqualinkD: RS received packet of type Status length 12
May-05-22 11:47:23 AM Debug:   PDA:       To 0x63 of type           Status | HEX: 0x10|0x02|0x63|0x02|0x00|0x00|0x00|0x00|0x00|0x77|0x10|0x03|
May-05-22 11:47:23 AM Debug:   AqualinkD: RS received packet of type Status length 12
May-05-22 11:47:23 AM Debug:   PDA:       To 0x63 of type           Status | HEX: 0x10|0x02|0x63|0x02|0x00|0x00|0x00|0x00|0x00|0x77|0x10|0x03|
May-05-22 11:47:23 AM Debug:   AqualinkD: RS received packet of type Status length 12
May-05-22 11:47:23 AM Debug:   PDA:       To 0x63 of type           Status | HEX: 0x10|0x02|0x63|0x02|0x00|0x00|0x00|0x00|0x00|0x77|0x10|0x03|
May-05-22 11:47:23 AM Debug:   AqualinkD: RS received packet of type Status length 12
May-05-22 11:47:23 AM Debug:   PDA:       To 0x63 of type           Status | HEX: 0x10|0x02|0x63|0x02|0x00|0x00|0x00|0x00|0x00|0x77|0x10|0x03|
May-05-22 11:47:23 AM Debug:   AqualinkD: RS received packet of type Status length 12
May-05-22 11:47:24 AM Debug:   PDA:       To 0x63 of type           Status | HEX: 0x10|0x02|0x63|0x02|0x00|0x00|0x00|0x00|0x00|0x77|0x10|0x03|
May-05-22 11:47:24 AM Debug:   AqualinkD: RS received packet of type Status length 12
May-05-22 11:47:24 AM Debug:   PDA:       To 0x63 of type           Status | HEX: 0x10|0x02|0x63|0x02|0x00|0x00|0x00|0x00|0x00|0x77|0x10|0x03|
May-05-22 11:47:24 AM Debug:   AqualinkD: RS received packet of type Status length 12
May-05-22 11:47:24 AM Debug:   PDA:       To 0x63 of type           Status | HEX: 0x10|0x02|0x63|0x02|0x00|0x00|0x00|0x00|0x00|0x77|0x10|0x03|
May-05-22 11:47:24 AM Debug:   AqualinkD: RS received packet of type Status length 12
May-05-22 11:47:24 AM Debug:   PDA:       To 0x63 of type           Status | HEX: 0x10|0x02|0x63|0x02|0x00|0x00|0x00|0x00|0x00|0x77|0x10|0x03|
May-05-22 11:47:24 AM Debug:   AqualinkD: RS received packet of type Status length 12
May-05-22 11:47:24 AM Debug:   PDA:       To 0x63 of type           Status | HEX: 0x10|0x02|0x63|0x02|0x00|0x00|0x00|0x00|0x00|0x77|0x10|0x03|
May-05-22 11:47:25 AM Debug:   AqualinkD: RS received packet of type Status length 12
May-05-22 11:47:25 AM Debug:   PDA:       To 0x63 of type           Status | HEX: 0x10|0x02|0x63|0x02|0x00|0x00|0x00|0x00|0x00|0x77|0x10|0x03|
May-05-22 11:47:25 AM Debug:   AqualinkD: RS received packet of type Status length 12
May-05-22 11:47:25 AM Debug:   PDA:       To 0x63 of type           Status | HEX: 0x10|0x02|0x63|0x02|0x00|0x00|0x00|0x00|0x00|0x77|0x10|0x03|
May-05-22 11:47:25 AM Debug:   AqualinkD: RS received packet of type Status length 12
May-05-22 11:47:25 AM Debug:   PDA:       To 0x63 of type           Status | HEX: 0x10|0x02|0x63|0x02|0x00|0x00|0x00|0x00|0x00|0x77|0x10|0x03|
May-05-22 11:47:25 AM Debug:   AqualinkD: RS received packet of type Status length 12
May-05-22 11:47:25 AM Debug:   PDA:       To 0x63 of type           Status | HEX: 0x10|0x02|0x63|0x02|0x00|0x00|0x00|0x00|0x00|0x77|0x10|0x03|
May-05-22 11:47:25 AM Debug:   AqualinkD: RS received packet of type Status length 12
May-05-22 11:47:25 AM Debug:   PDA:       To 0x63 of type           Status | HEX: 0x10|0x02|0x63|0x02|0x00|0x00|0x00|0x00|0x00|0x77|0x10|0x03|
May-05-22 11:47:25 AM Debug:   AqualinkD: RS received packet of type Status length 12
[... cut ...]
May-05-22 11:47:26 AM Debug:   PDA:       To 0x63 of type           Status | HEX: 0x10|0x02|0x63|0x02|0x00|0x00|0x00|0x00|0x00|0x77|0x10|0x03|
May-05-22 11:47:26 AM Debug:   AqualinkD: RS received packet of type Status length 12
May-05-22 11:47:26 AM Debug:   PDA:       To 0x63 of type           Status | HEX: 0x10|0x02|0x63|0x02|0x00|0x00|0x00|0x00|0x00|0x77|0x10|0x03|
May-05-22 11:47:26 AM Debug:   AqualinkD: RS received packet of type Status length 12
May-05-22 11:47:26 AM Debug:   PDA:       To 0x63 of type           Status | HEX: 0x10|0x02|0x63|0x02|0x00|0x00|0x00|0x00|0x00|0x77|0x10|0x03|

ballle98, 2022-05-05:
OK I think I may understand the issue, but I would have to go through the code to verify.  Your control panel sends a status message when I was not expecting it.  That is why it get retransmitted over and over again because it is not being ack'd

heffneil, 2022-05-05:
Ok let me know how I can be of assistance.  If you need more information about my panel or anything I am happy to help!

Thanks!

ballle98, 2022-05-05:
There is something that is still really off.  Like your running the wrong version of code.  So I can't debug it because logs that should be there are not there like:
` Debug:   PDA:       waitForPDAMessageTypesOrMenu`

n the directory where you did the git clone capture the output of these commands:

```
git remote -v
git log -1
sudo make install

```

follow these instructions to capture log https://github.com/ballle98/AqualinkD/wiki#run-in-foreground-with-logging

you can use putty logging to capture the terminal output on your pi

https://stackoverflow.com/questions/21231877/saving-the-putty-session-logging

heffneil, 2022-05-05:
Output:

jandy@jandypi:~/AqualinkD $ git remote -v
origin	https://github.com/ballle98/AqualinkD.git (fetch)
origin	https://github.com/ballle98/AqualinkD.git (push)
jandy@jandypi:~/AqualinkD $ git log -1
commit 12e537edf6c913999bd5a91e7b467fee85b81d4a (HEAD -> master, origin/master, origin/HEAD)
Author: Lee Ballard <ballle98@gmail.com>
Date:   Tue Mar 8 15:56:58 2022 -0600

    ballle98/AqualinkD#58: compiler warnings seen after bullseye update
jandy@jandypi:~/AqualinkD $ sudo make install
OS: 11 (bullseye)
GLIBC: ldd (Debian GLIBC 2.31-13+rpt2+rpi1+deb11u2) 2.31
./release/install.sh
AqualinkD config exists, did not copy new config, you may need to edit existing! /etc/aqualinkd.conf
AqualinkD defaults exists, did not copy new defaults to /etc/default/aqualinkd
Avahi/mDNS defaults exists, did not copy new defaults to /etc/avahi/services//aqualinkd.service
AqualinkD web config exists, did not copy new config, you may need to edit existing /var/www/aqualinkd//config.js
Starting daemon aqualinkd

ballle98, 2022-05-06:
OK try the foreground logging capture the log and you can attach it here as a gist
https://docs.github.com/en/github-ae@latest/get-started/writing-on-github/editing-and-sharing-content-with-gists/creating-gists

```
sudo systemctl stop aqualinkd
sudo release/aqualinkd -v -d -c /etc/aqualinkd.conf
```

You can see an example of my output from my pool here - [https://gist.github.com/ballle98/c641dd83d59072d4539fae50398a21e9](url)

heffneil, 2022-05-17:
I wanted to see if you are waiting for me for something?

ballle98, 2022-05-18:
I need updated logging collected like this

```
sudo systemctl stop aqualinkd
sudo release/aqualinkd -v -d -c /etc/aqualinkd.conf
```

I need the following messages to see what signal the code is waiting on.
```
14:41:37.661 Debug:   PDA:       waitForPDAMessageTypesOrMenu: received 0x09
14:41:37.662 Debug:   PDA:       waitForPDAMessageTypesOrMenu 0x08,0x10,0x02,(null),0,3.000 sec

```
The other thing that make no sense is there should be a log with an "Error:" indicating a timeout waiting on a signal.

heffneil, 2022-05-18:
09:50:16.339 Debug:   PDA:       waitForPDAnextMenu
09:50:16.339 Debug:   PDA:       waitForPDAMessageTypesOrMenu 0x09,0x02,0xff,(null),0,2.000 sec
09:50:16.368 Debug:   AqualinkD: RS received packet of type PDA Clear length 9
09:50:16.369 Debug:   PDA:       To 0x63 of type        PDA Clear | HEX: 0x10|0x02|0x63|0x09|0x00|0x00|0x7e|0x10|0x03|
09:50:16.369 Debug:   Panl Prog: Kicking PDA thread 10,0x19bf598
09:50:16.370 Debug:   PDA:       waitForPDAMessageTypesOrMenu: received 0x09
09:50:16.370 Debug:   PDA:       waitForPDAMessageTypesOrMenu 0x08,0x10,0x02,(null),0,3.000 sec
09:50:16.432 Debug:   AqualinkD: RS received packet of type Lng Message length 24
09:50:16.433 Debug:   PDA:       To 0x63 of type      Lng Message | HEX: 0x10|0x02|0x63|0x04|0x04|0x50|0x4f|0x4f|0x4c|0x20|0x4d|0x4f|0x44|0x45|0x20|0x20|0x20|0x20|0x20|0x4f|0x4e|0x39|0x10|0x03|
09:50:16.433 Debug:   PDA:       PDA Menu Line 0 =
09:50:16.433 Debug:   PDA:       PDA Menu Line 1 =
09:50:16.434 Debug:   PDA:       PDA Menu Line 2 =
09:50:16.434 Debug:   PDA:       PDA Menu Line 3 =
09:50:16.435 Debug:   PDA:       PDA Menu Line 4 = POOL MODE     ON
09:50:16.435 Debug:   PDA:       PDA Menu Line 5 =
09:50:16.436 Debug:   PDA:       PDA Menu Line 6 =
09:50:16.436 Debug:   PDA:       PDA Menu Line 7 =
09:50:16.436 Debug:   PDA:       PDA Menu Line 8 =
09:50:16.437 Debug:   PDA:       PDA Menu Line 9 =

heffneil, 2022-05-18:
09:50:49.042 Info:    RS Serial: Forcing close of blocking serial port, ignore following read errors
09:50:49.046 Warning: RS Serial: Read error: 9 - Bad file descriptor

ballle98, 2022-05-19:
Do you have a full log up to the point where it is getting the Equipment menu and there should be an Error: log reporting a timeout.  Then you see all these status messages.  I suspect the flood of status messages are re-transmissions from the pannel because aqualind did not ACK them because it is stuck waiting on another message. 

```
May-05-22 11:47:25 AM Debug: AqualinkD: RS received packet of type Status length 12
May-05-22 11:47:25 AM Debug: PDA: To 0x63 of type Status | HEX: 0x10|0x02|0x63|0x02|0x00|0x00|0x00|0x00|0x00|0x77|0x10|0x03|
```

heffneil, 2022-05-19:
Here is the whole file.
[output.txt](https://github.com/ballle98/AqualinkD/files/8729415/output.txt)

ballle98, 2022-05-19:
Got it thanks probably won't be able to dig into it before Monday.

ballle98, 2022-05-20:
The lalels in the config file must be an exact case insensitive match for the way things dispaly on the equipment menu.

"Spa Mode" shold be "Spa"
"Spa Blower" should be "Air Blower"
"Pool Heater" should be "Pool Heat"
"Spa Heather" should be Spa Heat"

If you use EXTRA AUX then set the label of AUX_6 or Aux_7 to "EXTRA AUX"

```
09:57:20.371 Notice:  AqualinkD: Config BTN Filter_Pump   = label Filter Pump     | equi
09:57:20.372 Notice:  AqualinkD: Config BTN Spa_Mode      = label Spa Mode        | 
09:57:20.372 Notice:  AqualinkD: Config BTN Aux_1         = label Cleaner         | 
09:57:20.372 Notice:  AqualinkD: Config BTN Aux_2         = label Waterfall       | 
09:57:20.372 Notice:  AqualinkD: Config BTN Aux_3         = label Spa Blower      | 
09:57:20.373 Notice:  AqualinkD: Config BTN Aux_4         = label Pool Light      | 
09:57:20.373 Notice:  AqualinkD: Config BTN Aux_5         = label Spa Light       | 
09:57:20.373 Notice:  AqualinkD: Config BTN Aux_6         = label NONE            | 
09:57:20.373 Notice:  AqualinkD: Config BTN Aux_7         = label NONE            | 
09:57:20.373 Notice:  AqualinkD: Config BTN Pool_Heater   = label Pool Heater     | 
09:57:20.374 Notice:  AqualinkD: Config BTN Spa_Heater    = label Spa Heater      | 
09:57:20.374 Notice:  AqualinkD: Config BTN Solar_Heater  = label Solar Heater    | 
```

```
09:57:28.584 Debug:   PDA:       PDA Menu Line 0 =    EQUIPMENT    
09:57:28.584 Debug:   PDA:       PDA Menu Line 1 = FILTER PUMP   ON
09:57:28.584 Debug:   PDA:       PDA Menu Line 2 = SPA          OFF
09:57:28.585 Debug:   PDA:       PDA Menu Line 3 = POOL HEAT    OFF
09:57:28.585 Debug:   PDA:       PDA Menu Line 4 = SPA HEAT     OFF
09:57:28.586 Debug:   PDA:       PDA Menu Line 5 = CLEANER      OFF
09:57:28.586 Debug:   PDA:       PDA Menu Line 6 = POOL LIGHT   OFF
09:57:28.586 Debug:   PDA:       PDA Menu Line 7 = SPA LIGHT    OFF
09:57:28.587 Debug:   PDA:       PDA Menu Line 8 = AIR BLOWER   OFF
09:57:28.587 Debug:   PDA:       PDA Menu Line 9 =    ^^ MORE __   

```

```
09:57:32.679 Debug:   PDA:       PDA Menu Line 0 =    EQUIPMENT    
09:57:32.679 Debug:   PDA:       PDA Menu Line 1 = SPA HEAT     OFF
09:57:32.680 Debug:   PDA:       PDA Menu Line 2 = CLEANER      OFF
09:57:32.680 Debug:   PDA:       PDA Menu Line 3 = POOL LIGHT   OFF
09:57:32.681 Debug:   PDA:       PDA Menu Line 4 = SPA LIGHT    OFF
09:57:32.681 Debug:   PDA:       PDA Menu Line 5 = AIR BLOWER   OFF
09:57:32.682 Debug:   PDA:       PDA Menu Line 6 = WATERFALL    OFF
09:57:32.682 Debug:   PDA:       PDA Menu Line 7 = EXTRA AUX    OFF
09:57:32.682 Debug:   PDA:       PDA Menu Line 8 = ALL OFF         
09:57:32.683 Debug:   PDA:       PDA Menu Line 9 =    ^^ MORE __   

```

ballle98, 2022-05-21:
It looks like the time is close enought that it will not try to sync the time.  If the day of the week is correct and the time is within 2 min it will not try to update the time.  If you want to validate it then change the day or set the time off by an hour and it will update it.

`09:57:26.472 Debug:   PDA:       PDA Menu Line 0 =      WED  9:56AM
`

`09:57:26.476 Debug:   PDA:       process_pda_packet_msg_long_time strptime Wed May 18 09:56:26 2022
`

You may want to uncomment "pda_sleep_mode = yes" your etc/aqualinkd.comf

```
# Put AqualinkD to sleep when in PDA mode after inactivity. 
# Ignore if you are not using PDA mode.
# If you have Jandy PDA then this MUST be set to yes as the controller can only support one PDA.
# If you don't have a Jandy PDA leave this at no as AqualinkD will be a lot quicker.
#pda_sleep_mode = yes
```

heffneil, 2022-05-21:
Ok  I have uncommented the sleep line now.

I am unclear on how the equipment menu should look.  I will research more directions now but if you have an example that would be great.

heffneil, 2022-05-21:
Ok I think I got that config sorted.  Not sure what to do about button 12?  i had PDA-8 but its PDA-6.

heffneil, 2022-05-21:
I changed those things and it still reflects running - when it not.

heffneil, 2022-05-23:
New output file - with updated settings.  I don't think its better.
[output.txt](https://github.com/ballle98/AqualinkD/files/8755141/output.txt)

ballle98, 2022-05-23:
Spa and pool heaters still look wrong

```
16:26:04.635 Notice:  AqualinkD: Config BTN Filter_Pump   = label Filter Pump     | 
16:26:04.635 Notice:  AqualinkD: Config BTN Spa_Mode      = label Spa             | 
16:26:04.635 Notice:  AqualinkD: Config BTN Aux_1         = label Cleaner         | 
16:26:04.636 Notice:  AqualinkD: Config BTN Aux_2         = label Waterfall       | 
16:26:04.636 Notice:  AqualinkD: Config BTN Aux_3         = label Air Blower      | 
16:26:04.636 Notice:  AqualinkD: Config BTN Aux_4         = label Pool Light      | 
16:26:04.636 Notice:  AqualinkD: Config BTN Aux_5         = label Spa Light       | 
16:26:04.636 Notice:  AqualinkD: Config BTN Pool_Heater   = label NONE            | 
16:26:04.637 Notice:  AqualinkD: Config BTN Spa_Heater    = label NONE            | 
16:26:04.637 Notice:  AqualinkD: Config BTN Solar_Heater  = label Pool Heat       | 

```

Edit these lines in your /etc/aqualinkd.conf

```
button_10_label=Pool Heat
button_11_label=Spa Heat
button_12_label=NONE
```

In the log files I see "FILTER PUMP   ON".  This is telling aqualinkd that the filter pump is running.  If you use you physical PDA is there another idicator?

```
16:29:46.055 Debug:   PDA:       PDA Menu Line 0 =    EQUIPMENT    
16:29:46.056 Debug:   PDA:       PDA Menu Line 1 = FILTER PUMP   ON
16:29:46.056 Debug:   PDA:       PDA Menu Line 2 = SPA          OFF
16:29:46.056 Debug:   PDA:       PDA Menu Line 3 = POOL HEAT    OFF
16:29:46.057 Debug:   PDA:       PDA Menu Line 4 = SPA HEAT     OFF
16:29:46.057 Debug:   PDA:       PDA Menu Line 5 = CLEANER      OFF
16:29:46.058 Debug:   PDA:       PDA Menu Line 6 = POOL LIGHT   OFF
16:29:46.058 Debug:   PDA:       PDA Menu Line 7 = SPA LIGHT    OFF
16:29:46.058 Debug:   PDA:       PDA Menu Line 8 = AIR BLOWER   OFF
```

heffneil, 2022-05-23:
Im re-running it to the output file and I see this on the command line:

Error:   PDA:       waitForPDAMessageTypesOrMenu 0x08,0x04,0xff,(null),0 - Connection timed out
Error:   PDA:       waitForPDAMessageTypesOrMenu: did not receive 0x08, 0x04 or 0xff
Error:   PDA:       PDA Device programmer request for menu 19, stuck on 2
Error:   PDA:       PDA Device programmer didn't find a requested menu 19, current 2
Error:   PDA:       loopover_devices :- can't goto PM_EQUIPTMENT_CONTROL menu
Error:   PDA:       PDA Wake Init :- can't find menu
Error:   PDA:       PDA Device programmer request for menu 19, stuck on 2
Error:   PDA:       PDA Device programmer didn't find a requested menu 19, current 2
Error:   PDA:       loopover_devices :- can't goto PM_EQUIPTMENT_CONTROL menu
Error:   PDA:       PDA Wake Init :- can't find menu
Error:   PDA:       PDA Device programmer request for menu 19, stuck on 2
Error:   PDA:       PDA Device programmer didn't find a requested menu 19, current 2
Error:   PDA:       loopover_devices :- can't goto PM_EQUIPTMENT_CONTROL menu
Error:   PDA:       PDA Wake Init :- can't find menu

ballle98, 2022-05-23:
Is the physical PDA turned on?  only one PDA device can talk to the control panel at a time.  Do you have a variable speed pump?  There are also settings for delays for example when you switch spa mode on the filter pump is delayed or when you turn the spa heater off the pump is hel on for some time.  If you replaced the pump there may be some miss communication.  The control panel will send commands to the pump over the serial.  It's just a different address than the PDA.

heffneil, 2022-05-23:
PDA is not turned on.  I rarely turn it on and when doing the output of this running it hasn't been turned on for days.  I do not have a variable speed pump.  I have a basic on / off pump nothing else there.

ballle98, 2022-05-23:
If you press the filter pump button on theweb gui and wait it does not turn off?  After you press off it can take 10-15 seconds to complete the action.

heffneil, 2022-05-23:
Yes I can control the pool pump.  It will turn on and off.  PDA - is never ever used.  I turned it on the other day to look at labels but had trouble connecting so turned it off.  Its cumbersome to use and never turned on unless we update programming.

heffneil, 2022-05-24:
who spelled equipment PM_EQUIPTMENT_CONTROL?

heffneil, 2022-05-24:
no matter what I do I can not get those buttons to be read from the config file...

ballle98, 2022-05-25:
Not sure if this works but you could try changing this to yes

```
# Use/find labels from Control Panel, these will overwrite the button_xx_label below, 
# it noes NOT work in PDA mode, and it also considerable slows down AqualinkD startup process.
use_panel_aux_labels=no
```

ballle98, 2022-05-25:
can you attach your config file and I'll see if I can reproduce the issue?

heffneil, 2022-05-25:
Here you go and thank you very very much!
[aqualink.txt](https://github.com/ballle98/AqualinkD/files/8773184/aqualink.txt)

heffneil, 2022-05-31:
Checking in - hopeful you found something I screwed up in the config file? Thanks!

ballle98, 2022-05-31:
Sorry got busy with other things.  Should have some time this week to look at it.

ballle98, 2022-06-01:
OK there is some issue with enabeling a log file in aqualinkd.com "log_file=/var/log/aqualinkd.log" When I commented out this line things started working much better.

The other thing was "panel_type = PD-6 Combo"  If you lokk at the notes in the conf file it changes the location of your pool heater to button 8 and spa heater to button 9.  Instead of fixing the labels change your panel type to "PD-8 Combo" because it will work better there seems to be some spots in the code where it is hard coded to  to 10 and 11.
[aqualink-fixed.txt](https://github.com/ballle98/AqualinkD/files/8819049/aqualink-fixed.txt)

ballle98, 2022-06-14:
@heffneil Did this fix your issue?  Are you OK to close?

heffneil, 2022-06-14:
Hi I am traveling right now and haven’t really tested I did remove the logging reference but that in itself is a big, no?

ballle98, 2022-06-15:
OK I updated the SWG code for PDA so you may want to pull down the latet code and make clean; make install.  I'm not sure what the issue with log file is.  It may be delaying the serial reader thread to the point it is missing messages or corrupting messages.  It definately caused some strange issues on my setup.  The panel type is another feature that is not complete for PDA mode.  I have a PD-4 Combo pool and I have to set it to PD-8 Combo to work properly.

heffneil, 2022-06-21:
It at appears as if it is responding swiftly turning things on and off

ballle98, 2022-06-21:
OK.  I'll close the issue :)

closed 2022-06-21 by ballle98