ballle98/AqualinkD #76: Just can't get this to work reliably

heffneil, 2023-06-14:
Im just getting inundated with this:

17:09:15.804 Debug:   PDA:       To 0x62 of type           Status | HEX: 0x10|0x02|0x62|0x02|0x00|0x00|0x00|0x00|0x00|0x76|0x10|0x03|
17:09:15.978 Debug:   AqualinkD: RS received packet of type Status length 12
17:09:15.980 Debug:   PDA:       To 0x62 of type           Status | HEX: 0x10|0x02|0x62|0x02|0x00|0x00|0x00|0x00|0x00|0x76|0x10|0x03|
17:09:16.154 Debug:   AqualinkD: RS received packet of type Status length 12
17:09:16.156 Debug:   PDA:       To 0x62 of type           Status | HEX: 0x10|0x02|0x62|0x02|0x00|0x00|0x00|0x00|0x00|0x76|0x10|0x03|
17:09:16.330 Debug:   AqualinkD: RS received packet of type Status length 12
17:09:16.332 Debug:   PDA:       To 0x62 of type           Status | HEX: 0x10|0x02|0x62|0x02|0x00|0x00|0x00|0x00|0x00|0x76|0x10|0x03|
17:09:16.506 Debug:   AqualinkD: RS received packet of type Status length 12
17:09:16.508 Debug:   PDA:       To 0x62 of type           Status | HEX: 0x10|0x02|0x62|0x02|0x00|0x00|0x00|0x00|0x00|0x76|0x10|0x03|
17:09:16.682 Debug:   AqualinkD: RS received packet of type Status length 12
17:09:16.684 Debug:   PDA:       To 0x62 of type           Status | HEX: 0x10|0x02|0x62|0x02|0x00|0x00|0x00|0x00|0x00|0x76|0x10|0x03|
17:09:16.858 Debug:   AqualinkD: RS received packet of type Status length 12
17:09:16.860 Debug:   PDA:       To 0x62 of type           Status | HEX: 0x10|0x02|0x62|0x02|0x00|0x00|0x00|0x00|0x00|0x76|0x10|0x03|
17:09:17.034 Debug:   AqualinkD: RS received packet of type Status length 12
17:09:17.036 Debug:   PDA:       To 0x62 of type           Status | HEX: 0x10|0x02|0x62|0x02|0x00|0x00|0x00|0x00|0x00|0x76|0x10|0x03|
17:09:17.210 Debug:   AqualinkD: RS received packet of type Status length 12
17:09:17.212 Debug:   PDA:       To 0x62 of type           Status | HEX: 0x10|0x02|0x62|0x02|0x00|0x00|0x00|0x00|0x00|0x76|0x10|0x03|
17:09:17.386 Debug:   AqualinkD: RS received packet of type Status length 12
17:09:17.388 Debug:   PDA:       To 0x62 of type           Status | HEX: 0x10|0x02|0x62|0x02|0x00|0x00|0x00|0x00|0x00|0x76|0x10|0x03|
17:09:17.562 Debug:   AqualinkD: RS received packet of type Status length 12
17:09:17.564 Debug:   PDA:       To 0x62 of type           Status | HEX: 0x10|0x02|0x62|0x02|0x00|0x00|0x00|0x00|0x00|0x76|0x10|0x03|

ballle98, 2023-06-15:
Interesting.  Would be good to get the full log and attach it to this issue as a txt file.  See https://github.com/ballle98/AqualinkD/wiki#run-in-foreground-with-logging

Do you have "pda_sleep_mode = yes" in /etc/aqualinkd.conf?

You are probably seeing the status command repeated because it is not being ACK'd.  This is expected behavior when aqualinkd is sleeping but your pannel may behave differently than mine.  After not ack'ing a few status commands the control panel should start probing. 

```
# Put AqualinkD to sleep when in PDA mode after inactivity. 
# Ignore if you are not using PDA mode.
# If you have Jandy PDA then this MUST be set to yes as the controller can only support one PDA.
# If you don't have a Jandy PDA leave this at no as AqualinkD will be a lot quicker.
## Put AqualinkD to sleep when in PDA mode after inactivity. 
# Ignore if you are not using PDA mode.
# If you have Jandy PDA then this MUST be set to yes as the controller can only support one PDA.
# If you don't have a Jandy PDA leave this at no as AqualinkD will be a lot quicker.
#pda_sleep_mode = yes
```

ballle98, 2023-06-16:
see https://github.com/sfeakes/AqualinkD/discussions/219

The issue seems to be related to service mode.  Honestly that should me supported though and show up in the logs so I'm still not clear what the issue is.

closed 2023-06-16 by ballle98